pixelmatix/aurora #10: Missing aJSON.h

RGB-123, 2015-01-10:
Louis, it looks like the #include "aJSON.h" is missing from this repo.  Can you please push this update?

pup05, 2015-01-10:
I see that include in Aurora.ino in the master branch: https://github.com/pixelmatix/aurora/blob/master/Aurora.ino#L51

RGB-123, 2015-01-10:
Right, the include is in there, but the actual file is not in the repo and hence I can't compile aurora.

pup05, 2015-01-10:
Yeah, you have to download it separately, just like all the other libraries (SmartMatrix, FastLED, etc).  It wasn't originally mentioned in the libraries list until recently.  See the Compiling page in the wiki for more info: https://github.com/pixelmatix/aurora/wiki/Compiling

RGB-123, 2015-01-10:
Got it.  I was looking at the readme.md and should've gone to the wiki.  Good work on the wiki BTW!

closed 2015-01-10 by RGB-123

pup05, 2015-01-10:
Cool, sorry for the trouble, that was something I missed and another user fixed.  :)  Thanks!

kigster, 2015-01-10:
@RGB-123 actually aJSON library is included in README also.