pixelmatix/aurora #45: Arduino/Esp8266/16x16matrix support?

80Stepko08, 2020-08-15:


kostyamat, 2020-08-15:
У вас же матрица 8х8, нет?

Пс. Автор до сих пор не ответил ни на одно иссуэ. Не думаю, что ответит и на ваше. Придется самим разбираться, наверное. :(

jasoncoon, 2020-08-15:
> You have an 8x8 matrix, no?
> 
> Ps. The author has not yet answered any of the Issue. I don’t think he will answer yours either. We'll have to figure it out ourselves, probably.

I've been away from RGB matrix development for a while, but others have adapted many of the Aurora patterns to ESP8266 and ESP32:

https://github.com/mrfaptastic/ESP32-RGB64x32MatrixPanel-I2S-DMA/tree/master/examples/AuroraDemo

80Stepko08, 2020-08-15:
> У вас же матрица 8х8, нет?
> 
> Пс. Автор до сих пор не ответил ни на одно иссуэ. Не думаю, что ответит и на ваше. Придется самим разбираться, наверное. :(

Я просто хотел спросить, потом если смогу приобрету.

closed 2024-01-02 by St3p40