pixelmatix/aurora #46: Non of the Teensies work!

flasher69, 2021-11-29:
Hi guys,
I was trying to get my Smartled shield V4 and V5 and my Teensy 3.2, Teensy 3.5, Teensy 4.1 to work with a LED matrix 64 x 64. So far has nothing worked  for me.
I've flashed the Aurora.hex into my Teensies and copied all the files on a SD card and inserted them in Teensies. Still no luck. I also have tried to compile the source code using Teensyduino (Arduino 1.8.16 / Teensyduino 1.55) but I get a lot of errors.

I just wanted to know is there a working hex file/instruction out there or am I just wasting my time. After weeks of trying I'm kind of exhusted but I don't want to give up, if there is a solution.

cheers

embedded-creations, 2021-11-29:
Aurora isn't maintained anymore, but Manny has gotten it to compile, see this thread:

https://community.pixelmatix.com/t/too-many-errors-compiling-aurora-sketch/728/1

> I also have tried to compile the source code using Teensyduino (Arduino 1.8.16 / Teensyduino 1.55) but I get a lot of errors.

Please post what you tried and the errors over at community.pixelmatix.com and I'll try to help

flasher69, 2021-11-29:
Thanx a lot dear Louis for your fast reply. I'll check the link you've sent me. I think Aurora is one of the best LED Matrix projects and it's a big shame, that it doesn't get maintained anymore :(

embedded-creations, 2021-11-30:
I agree, there's not really anything like Aurora out there - at least not that I've found.  I'm putting my energy into improving WLED (which runs on the ESP32 not Teensy) to have similar functionality to Aurora but also with WLED's WiFi support and lots of other features.

flasher69, 2021-11-30:
I love WLED and I use it most of the time for my projects but I was also looking for something similar easy to use for HUB75 matrix. I didn't think that it can get so confusing.

flasher69, 2021-11-30:
Btw great work on WLED and thanx for all your efforts Louis :)